MicrosoftDocs/windows-driver-docs-ddi #659: -2 doesn't always work..

rodwiddowson, 2019-10-16:
Lori, So very sorry about this...

It turns out that you can set -1 into the times for the file `\System Volume Information\tracking.log.tmp` but if you try to set it to -2 then you'll get back```
Error code: (NTSTATUS) 0xc000000d (3221225485) - An invalid parameter was passed to a service or function.
```
Do you care? Not sure I do, SVI is really pretty much "hands off".    Feel free to close this as "mistaken", or we can add some extra words,,,

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30d6a301-2028-bba5-0e43-092824810891
* Version Independent ID: 5c8c373a-b34a-5e73-8002-57076540e6ce
* Content: [_FILE_BASIC_INFORMATION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-gb/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_file_basic_information)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

lorihollasch, 2020-01-22:
@rodwiddowson , which file system are you running on? It turns out that only NTFS and ReFS support special value of -2. I'll update the page with that info ...

rodwiddowson, 2020-01-22:
@lorihollasch it was a while ago but if it was SVI then almost by definition it was NTFS, it might have been REFS but it wouldn’t have been one of the FATs or UDF (whatever happened to Pinball :-)...

I think in this case it is the SVI-ness which is causing the weirdness (SVI is very often weird) but I don’t *know*...

lorihollasch, 2020-02-13:
@rodwiddowson, finally got to circle back on this issue. I've updated the page to say that only NTFS and ReFS recognize and handle the -2 value. Craig indicated that SVI doesn't really mean anything to NTFS. Might a filter have gotten involved and failed it? Or are you running on an older version of Windows? (-2 was added in Win 10 Threshold) I can't do much more from a doc standpoint so gonna close this out.

closed 2020-02-13 by lorihollasch

rodwiddowson, 2020-02-13:
Pretty sure it was NTFS (the SVI is a giveaway), but Its a long time ago.  I'm also reasonably sure that it was clean of lower filters (apart from the standard Windows one), but I'll believe Craig..