MicrosoftDocs/windows-driver-docs-ddi #662: Incorrect statement. IoAcquireRemoveLock does not have a return type.

jmc5113, 2019-10-18:
IoAcquireRemoveLock returns STATUS_SUCCESS if the call was successful. One possible error return value is the following:

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aff43c16-5161-b97d-e498-bbc5d2659fba
* Version Independent ID: 26169519-6e0e-42b2-851e-d8fc67a1b03a
* Content: [IoAcquireRemoveLock macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioacquireremovelock#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-18:
Thanks for the feedback, Jay.  First a caveat that I'm not a dev.  Here's what I see in wdm.h:

```
#if DBG
#define IoAcquireRemoveLock(RemoveLock, Tag) \
        IoAcquireRemoveLockEx(RemoveLock, Tag, __FILE__, __LINE__, sizeof (IO_REMOVE_LOCK))
#else
#define IoAcquireRemoveLock(RemoveLock, Tag) \
        IoAcquireRemoveLockEx(RemoveLock, Tag, "", 1, sizeof (IO_REMOVE_LOCK))
#endif

#if (NTDDI_VERSION >= NTDDI_WIN2K)
_IRQL_requires_max_(DISPATCH_LEVEL)
NTKERNELAPI
NTSTATUS
NTAPI
IoAcquireRemoveLockEx (
    _Inout_ PIO_REMOVE_LOCK RemoveLock,
    _In_opt_ PVOID          Tag, // Optional
    _In_ PCSTR              File,
    _In_ ULONG              Line,
    _In_ ULONG              RemlockSize // are we checked or free
    );
#endif
```

Doesn't the `#define` just expand out at compile time?  So yes, the "macro" is technically void, as our build system correctly puts in the syntax block, but seems to me the text in the return value section is fine.  It's just a text replacement to a call that returns NTSTATUS.

Did you encounter a scenario where that wasn't the case?  Is your recommendation that we delete all text in the return value section?

tedhudek, 2019-10-18:
Agreed with your comment in #663 about adding NTSTATUS in the syntax block.

I just pushed a change that should override the build system and render that as the syntax block.  Give it 30-60 min to appear on the live site.  Feel free to reopen if you still see any issues at that point.

closed 2019-10-18 by tedhudek

jmc5113, 2019-10-18:
Thank you for updating the doc.

FYI, I’m a bluetooth driver dev at MS (jcourtn). In case we happen to work together in the future 😊

Closed #662<https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/662>.

tedhudek, 2019-10-18:
Sounds good, Jay, thanks again for the feedback.

Look fwd to seeing you down the road.