MicrosoftDocs/windows-driver-docs-ddi #663: Why talk about return values that a driver developer has no way to receive?

jmc5113, 2019-10-18:
Return Value
This macro wraps and assumes the return value of IoAcquireRemoveLockEx, which is NTSTATUS.

IoAcquireRemoveLock returns STATUS_SUCCESS if the call was successful. One possible error return value is the following:

STATUS_DELETE_PENDING: The driver has received an IRP_MN_REMOVE_DEVICE for the device and has called IoReleaseRemoveLockandWait. That routine is waiting for all remove locks to clear before returning control to the driver.

If the routine returns any value besides STATUS_SUCCESS, do not start any new operations on the device.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: aff43c16-5161-b97d-e498-bbc5d2659fba
* Version Independent ID: 26169519-6e0e-42b2-851e-d8fc67a1b03a
* Content: [IoAcquireRemoveLock macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioacquireremovelock)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioacquireremovelock.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-10-18:
Hi Jay, happy to work with you on this.  Looks like this is essentially a dup of #662 so closing this one out.

closed 2019-10-18 by tedhudek

jmc5113, 2019-10-18:
Thanks. Yes, it looks like the real issue is that the return type for IoAcquireRemoveLock should be NTSTATUS. I see there was an item for this about a year ago but the doc is still incorrect.

-Jay

Closed #663<https://github.com/MicrosoftDocs/windows-driver-docs-ddi/issues/663>.

tedhudek, 2019-10-18:
Do you have a link to the item you reference from a year ago?