MicrosoftDocs/windows-driver-docs-ddi #667: Details of the Parameters are missing

xiaoyinl, 2019-10-26:
This page on docs.microsoft.com doesn't show the details of the parameters except for `flowContext`. But the markdown source nc-fwpsk-fwps_callout_classify_fn2.md is correct though.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 666e04bc-2285-5159-4689-892e3dd8865b
* Version Independent ID: 5c851501-f8c0-876d-a52b-98e5aec41e2f
* Content: [FWPS_CALLOUT_CLASSIFY_FN2 (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nc-fwpsk-fwps_callout_classify_fn2#feedback)
* Content Source: [wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

xiaoyinl, 2019-10-26:
Correct in GitHub Preview:
[image]

But not correctly shown on Microsoft's document page:
[image]

duncanmacmichael, 2019-10-28:
Hi @xiaoyinl, good find! I believe this is the result of parameter names not matching what the back-end rendering system thinks they should be called - the asterisk is throwing it off. I've fixed this and the other two classify callback pages, and the changes should be live later today. Thanks!

closed 2019-10-28 by duncanmacmichael