MicrosoftDocs/windows-driver-docs-ddi #669: UdecxEndpointInit - confusing description

pankajbharti, 2019-10-27:
UdecxEndpointInit

A pointer to an UDECXUSBENDPOINT_INIT structure that the client driver retrieved in the previous call to UdecxUsbSimpleEndpointInitAllocate.

----
I am confused since this is a callback. this structure is being passed to the client. Why does the client need to allocate it using A pointer to an UDECXUSBENDPOINT_INIT structure that the client driver retrieved in the previous call to UdecxUsbSimpleEndpointInitAllocate. Is the documentation wrong? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 695d1244-a3d5-a6a4-edaf-df4d4fd55e42
* Version Independent ID: 7c43c0ef-fea3-b8be-ea28-b818555c724a
* Content: [EVT_UDECX_USB_DEVICE_DEFAULT_ENDPOINT_ADD (udecxusbdevice.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/udecxusbdevice/nc-udecxusbdevice-evt_udecx_usb_device_default_endpoint_add#feedback)
* Content Source: [wdk-ddi-src/content/udecxusbdevice/nc-udecxusbdevice-evt_udecx_usb_device_default_endpoint_add.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/udecxusbdevice/nc-udecxusbdevice-evt_udecx_usb_device_default_endpoint_add.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-10-28:
Hi @pankajbharti, good question. I can see why the description might confuse you, although it is accurate. I'll explain the best I can here. The UDECXUSBENDPOINT_INIT structure is indeed retrieved by the driver by calling [UdecxUsbSimpleEndpointInitAllocate](https://docs.microsoft.com/windows-hardware/drivers/ddi/udecxusbendpoint/nf-udecxusbendpoint-udecxusbsimpleendpointinitallocate), which takes place during the virtual device initialization steps. See the sample here: [Create a virtual USB device](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/writing-a-ude-client-driver#create-a-virtual-usb-device). Then, the driver typically stores this init object in the device context space before calling [UdecxUsbDeviceInitSetStateChangeCallbacks](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/udecxusbdevice/nf-udecxusbdevice-udecxusbdeviceinitsetstatechangecallbacks). So, because the framework now knows the address of the init structure, it can pass it back to the driver in the call to EvtUdecxUsbDeviceDefaultEndpointAdd. See the sample here for this step: [Create dynamic endpoints](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/writing-a-ude-client-driver#create-dynamic-endpoints). 

This is my understanding of it. Does this help?

billlattimer, 2020-11-06:
Closing as guidance was provided but no further response received.

closed 2020-11-06 by billlattimer