MicrosoftDocs/windows-driver-docs-ddi #673: IRQL not specified for WDI miniport entry points

udoe, 2019-10-30:
Hi,

The documentation does not specify the IRQL requirement for any of the entry points the miniport provides in NDIS_MINIPORT_DRIVER_WDI_CHARACTERISTICS. In order to implement a WDI miniport, this information is essential.

Thanks.
Udo

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9e3fa279-7b85-942c-8dbb-db25a062da6d
* Version Independent ID: ad946869-9c7b-aa9d-ec7f-265b86703c2d
* Content: [_NDIS_MINIPORT_DRIVER_WDI_CHARACTERISTICS (dot11wdi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/dot11wdi/ns-dot11wdi-_ndis_miniport_driver_wdi_characteristics)
* Content Source: [wdk-ddi-src/content/dot11wdi/ns-dot11wdi-_ndis_miniport_driver_wdi_characteristics.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dot11wdi/ns-dot11wdi-_ndis_miniport_driver_wdi_characteristics.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-10-31:
Hi @udoe, good question. I searched through the source headers, our internal code browsing tools, and the WDI spec but couldn't find any info about the IRQL for these. The only clue I could find was in the N6SdioWdi_AllocateAdapter function in the WDI driver sample, which calls the NdisMGetDeviceProperty, which itself requires PASSIVE_LEVEL. But, this isn't really a definitive answer.

Let me email some folks on the WDI team and see if I can get an answer for you.

udoe, 2019-11-11:
Hi @duncanmacmichael, Is there any news? We are currently developing a WDI driver for our customer and poor documentation of the entry points is puzzling us.

Meanwhile we are wondering if the old native 802.11 model would have been a better choice. However, the documentation clearly states that WDI should be used for new drivers:
https://docs.microsoft.com/en-us/previous-versions/windows/hardware/wireless/native-802-11-software-architecture

Udo

billlattimer, 2019-11-22:
@udoe I've taken responsibility for this and I'm working for an answer; understand that you're blocked.

udoe, 2019-11-25:
I'm glad to hear that. Meanwhile we found more doc issues but missing IRQL specification is the most critical one.

tedhudek, 2021-02-25:
Just opened an internal work item to request IRQL annotations be added to dot11wdi.h.  Sorry for the delay!

closed 2021-02-25 by tedhudek

tedhudek, 2021-03-09:
Hi @udoe, couple extra points on this:

WIFICX based drivers are replacing WDI - to match more closely to the WDF model. (NDIS_MINIPORT_DRIVER_WDI_CHARACTERISTICS is not part of WIFICX.)

In general, we recommend moving to WIFICX, and don't invest on WDI anymore.

That said, as a general rule, for WDI - most require passive IRQL except for Direct OIDs.