MicrosoftDocs/windows-driver-docs-ddi #680: Data alignment

TomaszNowakSanDiego, 2019-11-06:
Data inside of this structure is 16-bytes aligned. Pretty important if someone would like to allocate buffer for this structure based on Maximum name size and Maximum data size.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e06bff35-5eb2-3ad8-9b15-651784757c60
* Version Independent ID: 4e31ec92-9346-a575-f58c-fc57c59832e6
* Content: [_KEY_VALUE_FULL_INFORMATION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_key_value_full_information)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_key_value_full_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_key_value_full_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

TomaszNowakSanDiego, 2019-11-06:
I'm sorry, looks like it's 8-bytes align.

tedhudek, 2019-11-06:
Thanks Tomasz.  Where would you suggest adding that info on the page?  Feel free to submit a PR if you have time to do so :)

(just click Edit from the docs page and it drops you into GitHub)

alfaunits, 2020-04-27:
> I'm sorry, looks like it's 8-bytes align.

Are you referring to member alignment, or size alignment?

TomaszNowakSanDiego, 2020-04-27:
There's no "Data" member in this structure, it is pointed by DataOffset. And that Data block must be 8-byte aligned.

tedhudek, 2020-07-17:
Sorry for the delay. @TomaszNowakSanDiego are you saying that the DataOffset member description needs language about being 8-byte aligned, or the KEY_VALUE_FULL_INFORMATION struct as a whole?  Apologies, I'm not familiar with this struct at all

tedhudek, 2020-07-22:
Closing this out due to inactivity - please reopen with info above as needed.

closed 2020-07-22 by tedhudek