MicrosoftDocs/windows-driver-docs-ddi #692: Typo in parameters name

rodwiddowson, 2019-11-22:
&gt; Parameters
&gt; 
&gt; Address of the caller's I/O status block. This parameter is required and cannot be NULL.

Parameter block, not IOSB.  

Some documentation of the parameters would be great (but an act of supererogation)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d54580f8-53ef-db25-71e1-dfccd8bfff5e
* Version Independent ID: ce8ef8a9-708b-a915-9bb6-1004e1842ec2
* Content: [NtFlushBuffersFileEx function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntflushbuffersfileex)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntflushbuffersfileex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntflushbuffersfileex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2019-11-22:
Also, it can be null because thats what happens when you call `NtFlushBuffersFile`

rodwiddowson, 2019-11-22:
Also
> If the call to the NtFlushBuffersFileEx function occurs in user mode, you should use the name "NtFlushBuffersFileEx" instead of "NtFlushBuffersFileEx".

And from looking at it in windbg the parameter must be NULL and the parameter size must be zero

lorihollasch, 2019-12-02:
@rodwiddowson , thanks for flagging this. Turns out that Parameters is currently not used, and that you are correct about NULL and zero settings. I've updated the page accordingly.

closed 2019-12-02 by lorihollasch