MicrosoftDocs/windows-driver-docs-ddi #693: Bugcheck code 0000007E

kouzhudong, 2019-11-27:
[Enter feedback here]
如果另一个驱动，知道RegistrationHandle，并且调用了ObUnRegisterCallbacks。
自己的驱动再调用RegistrationHandle会蓝屏。

2: kd> .bugcheck
Bugcheck code 0000007E
Arguments ffffffff`c0000005 fffff800`50171f43 ffffeb83`fe1db858 ffffeb83`fe1db0a0
2: kd> k
  *** Stack trace for last set context - .thread/.cxr resets it
 # Child-SP          RetAddr           Call Site
00 ffffeb83`fe1dba90 fffff800`50899fe2 nt!ExAcquirePushLockExclusiveEx+0x103
01 ffffeb83`fe1dbad0 fffff800`51e41866 nt!ObUnRegisterCallbacks+0x52
02 ffffeb83`fe1dbb00 fffff800`506e5ead test!Unload+0x16 [D:\test\ObRegisterCallbacks\test\test.c @ 237] 
03 ffffeb83`fe1dbb30 fffff800`5011211a nt!IopLoadUnloadDriver+0x8d
04 ffffeb83`fe1dbb70 fffff800`501d66c5 nt!ExpWorkerThread+0x16a
05 ffffeb83`fe1dbc10 fffff800`5026d49c nt!PspSystemThreadStartup+0x55
06 ffffeb83`fe1dbc60 00000000`00000000 nt!KiStartSystemThread+0x1c
2: kd> .cxr 0xffffeb83fe1db0a0
rax=0000000000000000 rbx=ffffc8845dc63040 rcx=ffffc8845dc63040
rdx=00000000000000b8 rsi=00000000000000b8 rdi=ffffc8845dc63360
rip=fffff80050171f43 rsp=ffffeb83fe1dba90 rbp=fffff800506e5e20
 r8=ffffeb83fe1dbad8  r9=0000000000000000 r10=fffff80051e41850
r11=0000000000000000 r12=0000000000000200 r13=0000000000000000
r14=ffffeb83fecdf750 r15=fffff80050594240
iopl=0         nv up ei ng nz na po nc
cs=0010  ss=0018  ds=002b  es=002b  fs=0053  gs=002b             efl=00010286
nt!ExAcquirePushLockExclusiveEx+0x103:
fffff800`50171f43 f0480fba2e00    lock bts qword ptr [rsi],0 ds:002b:00000000`000000b8=????????????????

这是谁的错呢？还是你微软没有错，用户用错了你的接口？

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5547887f-d270-a477-4012-9c71415c47fb
* Version Independent ID: a2364938-0b4e-dec6-5f8c-3b6e4cd147d5
* Content: [ObUnRegisterCallbacks function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-obunregistercallbacks)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-obunregistercallbacks.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-obunregistercallbacks.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-11-27:
Not clear what you feedback is.  Can you clarify?

kouzhudong, 2019-11-28:
If another driver knows the RegistrationHandle and calls it with ObUnRegisterCallbacks.
THe own driver once again call ObUnRegisterCallbacks and it will blue screen.

2: kd> .bugcheck
Bugcheck code 0000007E
Arguments ffffffffc0000005 fffff80050171f43 ffffeb83fe1db858 ffffeb83fe1db0a0

https://correy.webs.com

tedhudek, 2020-07-20:
Thanks @kouzhudong I confirmed that calling this function twice is the problem. It doesn't matter if it's the same driver or different drivers, either way you will get a bug check. I added a note to the doc page. The update should be live within the hour. Apologies for the delay and thanks for the feedback.

closed 2020-07-20 by tedhudek