MicrosoftDocs/windows-driver-docs-ddi #695: 64 Logical processors belong to different group

ivellioscolin, 2019-12-03:
The remarks below may need more revision.
"If a multiprocessor system is running a 64-bit version of Windows and contains no more than 64 logical processors, Windows assigns all processors to group 0."

On system with []sockets []cores []logical processors, set /NUMPROC to 64 and boot into Windows Server 2019. Now I got 2 sockets / 32 cores / 64 logical processors.

KeQueryActiveGroupCount() returns 2 and KeQueryActiveProcessorCountEx() returns 48  in group 0 and 16 in group 1. Same output from coreinfo.

So it's not true that all 64 logical processors belong to group 0.

Coreinfo output: (I suppressed core 1-30).

Logical to Physical Processor Map:
Physical Processor 0 (Hyperthreaded):
**----------------------------------------------
----------------
................
Physical Processor 31 (Hyperthreaded):
------------------------------------------------
--------------**

Logical Processor to Socket Map:
Socket 0:
************************************************
----------------
Socket 1:
------------------------------------------------
****************

Logical Processor to NUMA Node Map:
NUMA Node 0:
************************************************
----------------
NUMA Node 1:
------------------------------------------------
****************

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b286adc5-382a-edf8-c073-6d79084d80b7
* Version Independent ID: ca984721-3640-2758-4273-692fb6a525f9
* Content: [KeQueryActiveGroupCount function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-kequeryactivegroupcount#feedback)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-kequeryactivegroupcount.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-kequeryactivegroupcount.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-07-22:
Hi @ivellioscolin and apologies for the delay.

I was unable to get more info about this but I did just push a change to remove the sentence that you cite above.  If you have any further changes you would like to see there please feel free to open a PR (start by clicking **Edit** on the doc page).

Thanks again for reaching out!

closed 2020-07-22 by tedhudek

ivellioscolin, 2020-08-04:
Great! No confusion after the change.

tedhudek, 2020-08-05:
Had confirmation today that the situation you describe is normal, so removing that sentence was the correct step. Thanks for the sharp eye

ivellioscolin, 2020-08-12:
NP. MSDN is always the best Win document library. I'm glad to have helped a little.