MicrosoftDocs/windows-driver-docs-ddi #713: Incorrect link for additional reset info

I-gor-C, 2019-12-29:
The page states:

> For more information, see [Hardware Reset](https://docs.microsoft.com/previous-versions/windows/hardware/network/ff546572(v=vs.85)).

This link is both outdated (for NDIS 5.1) and also refers explicitly to 802.11 drivers.
This link is later reiterated in the Remarks section.

Also, missing on this page information regarding valid completion codes (I assume they should be the same as in [MINIPORT_RESET callback function](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ndis/nc-ndis-miniport_reset) but this should be stated explicitly).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c477b16d-d89b-3197-cfc9-88ae4094e4a5
* Version Independent ID: b859f178-4c09-ead2-b596-fb277a753dbe
* Content: [NdisMResetComplete macro (ndis.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ndis/nf-ndis-ndismresetcomplete?source=docs)
* Content Source: [wdk-ddi-src/content/ndis/nf-ndis-ndismresetcomplete.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ndis/nf-ndis-ndismresetcomplete.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-02-26:
Just pushed the link fix

tedhudek, 2021-03-02:
Thanks again for the feedback!  Fix is in above commit

closed 2021-03-02 by tedhudek

tedhudek, 2021-03-02:
One more thing, @I-gor-C --

Also note that **NdisMResetComplete** is two things - it's a macro in NDIS 5.x and a function in NDIS 6.x (for Vista and later).  I am currently updating the old page and adding a new page.  You'll see those updates on the Docs site soon.