MicrosoftDocs/windows-driver-docs-ddi #720: SetRenameInformationFile?

DaveInCaz, 2020-01-10:
I can to this page (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_rename_information) by searching on Google or docs.microsoft.com for "SetRenameInformationFile". But there is no mention of it on this page.

Nor is there any other page in docs... that refers to it. Seems like an omission?

Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7458144c-41ea-41a8-be0b-a7e8a786105b
* Version Independent ID: 17c91e00-d2a1-8984-30b3-db44d430f0f4
* Content: [_FILE_RENAME_INFORMATION (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_rename_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_rename_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_rename_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-01-10:
@daveincaz, thanks for reporting this. I've searched the source Markdown files (for FILE_RENAME_INFORMATION and every other ntifs.h source refpage file) and find no mention of SetRenameInformationFile. I searched the Windows source code, and can find one mention only (associated with Procmon). I can only guess that this page is coming up as a search engine result because it contains "Set", "Rename", "Information", and "File"? I'm going to close this since I can't pinpoint any missing docs - feel free to reach out if you can better determine what is missing.

closed 2020-01-10 by lorihollasch

DaveInCaz, 2020-01-10:
It was indeed Procmon which produced the name "SetRenameInformationFile" which I was searching for.

I guess the hypothetical missing information would be related to what that actually means when procmon prints it out. I had thought it would be an API / function call but obviously not.

Thanks for looking into it.

lorihollasch, 2020-01-10:
Not sure if this helps, but I found a link to [Procmon documentation](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon), which has a couple of links to additional info.