MicrosoftDocs/windows-driver-docs-ddi #721: Can we provide a documented guarantee that the Params.IoStatus field is always valid?

rahul-ramadas, 2020-01-14:
The page says:

> Note that the completion parameters structure contains valid information only if the driver has formatted the request by calling one of the WdfIoTargetFormatXxx methods.

However, WDF requests can also be formatted using `WdfRequestWdmFormatUsingStackLocation`, in which case I believe the `IoStatus` field in the completion params is valid (and only that field). This should be documented. Perhaps we should say that field is always valid.

@Shyamal-V FYI.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fb9e1e59-835a-5619-30ad-ea16c001bf6c
* Version Independent ID: 5e688c84-1338-f55d-c579-028043e62d18
* Content: [EVT_WDF_REQUEST_COMPLETION_ROUTINE (wdfrequest.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfrequest/nc-wdfrequest-evt_wdf_request_completion_routine)
* Content Source: [wdk-ddi-src/content/wdfrequest/nc-wdfrequest-evt_wdf_request_completion_routine.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfrequest/nc-wdfrequest-evt_wdf_request_completion_routine.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-20:
Thanks for the feedback @rahul-ramadas and apologies for the delay.

Spun up #874 to address.  Could you and @Shyamal-V please review?

tedhudek, 2020-07-02:
Thanks for getting the ball rolling on this @rahul-ramadas 

I just merged the fix - should be on the live site within the hour.

closed 2020-07-02 by tedhudek