MicrosoftDocs/windows-driver-docs-ddi #725: erroneous KeyValuePartialInformationAlign64 description

PisarevSergey, 2020-01-25:
For KeyValuePartialInformationAlign64 data stored not as KEY_VALUE_PARTIAL_INFORMATION but as KEY_VALUE_PARTIAL_INFORMATION_ALIGN64 .

Checked on Windows 10 x64 with latest updates.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f7587441-380b-2f29-32fc-3349703ec879
* Version Independent ID: ec04ec02-e153-9071-8bd2-b110b56f7964
* Content: [_KEY_VALUE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_key_value_information_class#feedback)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_key_value_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_key_value_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-19:
Thanks, just pushed the fix.

I wonder if KeyValueFullInformationAlign64 would then be stored as KEY_VALUE_FULL_INFORMATION_ALIGN64?

tedhudek, 2020-07-22:
@PisarevSergey can you confirm if the description for KeyValueFullInformationAlign64 should read "The information is stored as a KEY_VALUE_FULL_INFORMATION_ALIGN64 structure"?

PisarevSergey, 2020-07-22:
TL;DR: looks like current state of documentation about KeyValueFullInformationAlign64 return structure is correct and shouldn't be changed.

Long answer:
I am not aware about existence of KEY_VALUE_FULL_INFORMATION_ALIGN64 type.
Such type is not present in WDK includes (unlike KEY_VALUE_PARTIAL_INFORMATION_ALIGN64, that presents in the includes).

I tried to call NtQueryValueKey(user-mode gate to ZwQueryValueKey) with KeyValueFullInformationAlign64 and it looks like this call indeed returns correctly formatted KEY_VALUE_FULL_INFORMATION (not _ALIGN64) structure.
Checked on Windows 7 x64.

However it would be much better if someone from the inside of Microsoft who has access to source code confirmed that.

tedhudek, 2020-08-03:
Verified. Thanks Sergey!

closed 2020-08-03 by tedhudek