MicrosoftDocs/windows-driver-docs-ddi #726: Wrong alignment information for KeyValuePartialInformationAlign64

PisarevSergey, 2020-01-25:
Information won't be stored starting at the first 64-bit boundary in the buffer if buffer is not aligned.
Instead NtQueryValueKey returns status STATUS_DATATYPE_MISALIGNMENT.

Checked on Windows 10 x64 with latest updates.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f7587441-380b-2f29-32fc-3349703ec879
* Version Independent ID: ec04ec02-e153-9071-8bd2-b110b56f7964
* Content: [_KEY_VALUE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_key_value_information_class#feedback)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_key_value_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_key_value_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

PisarevSergey, 2020-01-25:
Same is true for KeyValueFullInformationAlign64

PisarevSergey, 2020-01-27:
NtQueryValueKey returns status STATUS_DATATYPE_MISALIGNMENT if (KeyValueInformation buffer is {aligned_address + 3});
aligned_address returned by malloc/operator new;

if KeyValueInformation buffer is {aligned_address + 4} everything works fine, but data in buffer is not offsetted on a 64-bit boundary, KeyValueInformation points right to KEY_VALUE_PARTIAL_INFORMATION_ALIGN64

tedhudek, 2020-06-19:
@PisarevSergey I started #873 to address your feedback - sorry this is delayed.

Would you take a look at the PR and add to it if you feel more info is needed?

PisarevSergey, 2020-06-21:
I did commented on the issue

closed 2020-06-22 by tedhudek