MicrosoftDocs/windows-driver-docs-ddi #734: Broken links

englishb, 2020-02-04:
Both hyperlinks to the D3DKMT_OPENADAPTERFROMDEVICENAME structure are broken links.  The search results on the 404 page also have a broken link for this structure.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bc8794be-4e38-c8a3-3e66-96d03631c2a0
* Version Independent ID: d8e667e0-0f41-d136-d44f-1954da52dc7b
* Content: [D3DKMTOpenAdapterFromDeviceName function (d3dkmthk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/nf-d3dkmthk-d3dkmtopenadapterfromdevicename)
* Content Source: [wdk-ddi-src/content/d3dkmthk/nf-d3dkmthk-d3dkmtopenadapterfromdevicename.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dkmthk/nf-d3dkmthk-d3dkmtopenadapterfromdevicename.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

englishb, 2020-02-04:
The google results for the structure are also broken links...perhaps the page has been deleted?

englishb, 2020-02-04:
And what seemed like mere seconds after I opened this issue, the page I opened the issue for is itself missing now...

englishb, 2020-02-04:
The pages for the D3DKMTOpenAdapterFromLuid function and related structure, as well as D3DKMTOpenAdapterFromHdc function and related structure are all missing now as well.

EliotSeattle, 2020-02-04:
@tedhudek I repro the missing page. Can you look into this ASAP or pass it on to Dave?

tedhudek, 2020-02-04:
Thanks for the report @englishb.  Turns out this is a bug on the publishing back end.  We're looking into it.

englishb, 2020-02-04:
Thank you.  I'm curious, if I had not opened this issue using the link on the page before it went down, how would I open an issue about a page being missing?

tedhudek, 2020-02-04:
You can just open an issue in this repo, which you should be able to get to by clicking Edit on any of the surviving pages on docs.microsoft.com.

englishb, 2020-02-04:
Good to know, thanks.

tedhudek, 2020-02-04:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/d3dkmthk/nf-d3dkmthk-d3dkmtopenadapterfromdevicename is back.  I don't repro the original issue - maybe it was related to the issue that was fixed?

englishb, 2020-02-04:
Yes, perhaps so.  All of the ones I mentioned seem to be working now.

tedhudek, 2020-02-04:
Apologies for the unpredictability, Ben, and thanks again for reaching out!

closed 2020-02-04 by tedhudek