MicrosoftDocs/windows-driver-docs-ddi #736: FsRtlMdlReadDev page actually describes FsRtlMdlRead

Mattiwatti, 2020-02-04:
The page whose link and URL describe FsRtlMdlRead**Dev** seems to consist of some combination of information about `FsRtlMdlRead` and `FsRtlMdlReadDev`. For example, the introduction and the Remarks section call the function 'FsRtlMdlRead' and make no mention of the device object parameter, but the prototype given is that for `FsRtlMdlReadDev` (with `PDEVICE_OBJECT DeviceObject` parameter).

Additionally, the `DeviceObject` parameter **is** listed in the Parameters section, but has a description that seems to have been copied from the description of the `IoStatus` parameter and therefore doesn't make any sense, because a device object is not an I/O status block.

I don't know which of the two functions this page is meant to describe, but it should either be fixed to be one or the other, or split into two separate pages for `FsRtlMdlRead` and `FsRtlMdlReadDev`.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b0770a4a-0649-17a6-2c04-22a2eb835491
* Version Independent ID: a8cde675-26db-415e-8048-b7c3d13a4cd6
* Content: [FsRtlMdlReadDev function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtlmdlreaddev)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtlmdlreaddev.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-_fsrtl_advanced_fcb_header-fsrtlmdlreaddev.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

hsebs, 2020-02-05:
The old location is wrong. That's why it's wrong.

It describes `FltFastIoMdlRead`, not `FsRtlMdlRead`

By the way, `FsRtlMdlRead` is deprecated in Windows 8.
If you want to use it, use `FsRtlMdlReadEx` instead of `FsRtlMdlRead`.

lorihollasch, 2020-05-22:
Thanks, @hsebs, for your PR that fixed this issue!

closed 2020-05-22 by lorihollasch