MicrosoftDocs/windows-driver-docs-ddi #740: Link to  FSRTL_CHANGE_BACKING_TYPE is out of date

rodwiddowson, 2020-02-07:
Lori (or whoever)
Apologies that I have no time to do a PR.  The link goes to the "unmaintained" pages rather that to [here|https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ne-ntifs-_fsrtl_change_backing_type]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f78c4a09-9257-e3f2-1ba0-a970eb010328
* Version Independent ID: 0029a0b0-b821-282f-1a9f-e63d3fd03e77
* Content: [FsRtlChangeBackingFileObject function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-fsrtlchangebackingfileobject)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlchangebackingfileobject.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-fsrtlchangebackingfileobject.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2020-02-07:
Working on it.

EliotSeattle, 2020-02-07:
Fixed and PR created. Should be live later today.

closed 2020-02-07 by EliotSeattle