MicrosoftDocs/windows-driver-docs-ddi #742: Wrong FILE_INFORMATION_CLASS on MSDN

0xeb, 2020-02-13:
The FILE_INFORMATION_CLASS enum on MSDN is 0-based. It should be instead:

```
typedef enum _FILE_INFORMATION_CLASS {
    FileDirectoryInformation         = 1,
    FileFullDirectoryInformation,   // 2
    FileBothDirectoryInformation,   // 3
    FileBasicInformation,           // 4
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a8119f8b-9c83-60f2-ac36-0478b88c4ab0
* Version Independent ID: 404ee010-b7dc-6a9e-7bbe-7cee72809333
* Content: [_FILE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_file_information_class)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-19:
Sorry for the delay on this.  This is a known issue with our doc reflection system. I just pushed a change that attempts to override the auto-gen syntax block, we'll see if it works.

tedhudek, 2020-07-22:
Fixed on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_file_information_class.  Thanks again for the report @0xeb

closed 2020-07-22 by tedhudek