MicrosoftDocs/windows-driver-docs-ddi #744: Not found by index in Help viewer 2.3

PisarevSergey, 2020-02-18:
Not found by index in Help viewer 2.3 but can be found by search

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f589015f-22d2-45b3-dee0-9da442a8b51f
* Version Independent ID: d12eee18-ba72-8a94-cd5f-578431697c52
* Content: [MmCopyMemory function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-mmcopymemory#feedback)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-mmcopymemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-mmcopymemory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-03-17:
Hi @PisarevSergey, it looks like that page was added in 2018.  In the Help Viewer, if you have the "Windows Driver Development" modules, those are from 2017.  But the "Windows Driver Kit(WDK)" modules are from 2019, and do contain that page.

[image]

tedhudek, 2020-03-18:
I had misunderstood.  I do repro that it is not in the index, but can be found using the Help Viewer search.  Comparing the metadata in the .md between that page and, say, IoCallDriver, I don't see a difference that would explain that.  Let me see if I can find anyone who might know why we're seeing this.

tedhudek, 2020-03-18:
It turns out that the whole index is from the "Windows Driver Kit(WDK)" module, and that the "Windows Driver Kit(WDK) Reference" module has no index.  The fix is to rebuild that book and republish it.  Thanks for alerting us to this, @PisarevSergey.  I'll update this issue when the new book is available.

tedhudek, 2020-03-23:
This is now fixed @PisarevSergey.  If you go into the Help Viewer, you'll see updates available for the WDK books.  That will fix the index in the Ref.  Thanks again for letting us know about the problem!

closed 2020-03-23 by tedhudek