MicrosoftDocs/windows-driver-docs-ddi #760: Is there equivalent methods for Universal platform as LoadLibrary to load KMDF coinstaller?

guliteb, 2020-03-05:
WdfPreDeviceInstallEx is marked as targeting Universal platform. However, LoadLibrary() only supports Desktop platform.

Is there equivalent methods for Universal platform as LoadLibrary to load KMDF coinstaller?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: df54170d-d7eb-e83a-1bea-1428063bd329
* Version Independent ID: f7ebdef3-8cc3-058d-2184-0d7b51764843
* Content: [WdfPreDeviceInstallEx function (wdfinstaller.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfinstaller/nf-wdfinstaller-wdfpredeviceinstallex)
* Content Source: [wdk-ddi-src/content/wdfinstaller/nf-wdfinstaller-wdfpredeviceinstallex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfinstaller/nf-wdfinstaller-wdfpredeviceinstallex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-03-05:
Which OS version(s) will your driver support?

guliteb, 2020-03-09:
I am working to make a pure Universal project. It includes driver/service/UWP. Hopefully to be compliant to Win10 S-mode and future universal Win10.

I ran APIvalidator on my console app with LoadLibrary and it didn't report error on LoadLibrary. It seems it is Universal. However, I ran APIvalidator on WdfCoInstaller01011.dll and it reports dependency on several SetupApi functions and result is not-universal.

Is WdfCoInstaller01011.dll really Universal?

tedhudek, 2020-03-09:
So will the driver only support Win10 and later?

guliteb, 2020-03-10:
Yes.

tedhudek, 2020-03-10:
Hi @guliteb, to support Win10 and later, we recommend using KMDF 1.15 without coinstaller. You should not need to reference WdfCoInstaller01011.dll from a universal driver.

guliteb, 2020-03-12:
@tedhudek , I am looking for a way to install/load non-PnP driver using an installer application. From the docs, it is recommended to use wdfcoinstaller to install non-PnP driver. Is there a new way in KMDF 1.15 to install non-PnP driver without using wdfcoinstaller?

This is the link I refer to:
[https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/installing-a-non-pnp-driver](url)

tedhudek, 2020-03-12:
You should be able to use the same approach as https://github.com/Microsoft/Windows-driver-samples/tree/master/general/ioctl/kmdf, but simply remove all the co-installer references, for example:

```
#define NONPNP_INF_FILENAME  L"\\nonpnp.inf"
#define WDF_SECTION_NAME L"nonpnp.NT.Wdf"
 
LoadWdfCoInstaller
UnloadWdfCoInstaller
 
PFN_WDFPREDEVICEINSTALLEX pfnWdfPreDeviceInstallEx;
PFN_WDFPOSTDEVICEINSTALL   pfnWdfPostDeviceInstall;
PFN_WDFPREDEVICEREMOVE     pfnWdfPreDeviceRemove;
PFN_WDFPOSTDEVICEREMOVE   pfnWdfPostDeviceRemove;
```

For non-pnp KMDF driver, simply call SCM API to create service is good enough.

I'll update https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/installing-a-non-pnp-driver with this info soon.

closed 2020-03-13 by tedhudek