MicrosoftDocs/windows-driver-docs-ddi #764: Incorrect size calculation for CM_PARTIAL_RESOURCE_LIST

tkreuzer, 2020-03-07:
According to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_partial_resource_descriptor, a CM_PARTIAL_RESOURCE_DESCRIPTOR of type CmResourceTypeDeviceSpecific is of variable size. According to a comment in wdm.h,  "There can only be *1* DeviceSpecificData block. It must be located at the end of all resource descriptors in a full descriptor block."

This implies that while indexing each partial resource descriptor within the CM_PARTIAL_RESOURCE_LIST works, the partial list's full size can not be calculated by the formula 
sizeof(CM_PARTIAL_RESOURCE_LIST) + (Count - 1) * sizeof(CM_PARTIAL_RESOURCE_DESCRIPTOR)
as described on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_partial_resource_list. Instead the calculation would be something like this:

ULONG SizeOfPartialResourceList(PCM_PARTIAL_RESOURCE_LIST PartialResourceList)
{
    ULONG Size = sizeof(CM_PARTIAL_RESOURCE_LIST) + 
        (PartialResourceList->Count - 1) * sizeof(CM_PARTIAL_RESOURCE_DESCRIPTOR);
    PCM_PARTIAL_RESOURCE_DESCRIPTOR LastDescriptor = 
        &PartialResourceList->PartialDescriptors[PartialResourceList->Count - 1];
    if (LastDescriptor ->Type ==  CmResourceTypeDeviceSpecific)
    {
        Size += LastDescriptor->u.DeviceSpecificData.DataSize;
    }
    return Size;
}

This applies also to the example code provided in https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_resource_list.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a8312daa-102b-a63f-816e-623523b9e171
* Version Independent ID: eea226ec-e1df-0f57-208d-0db41d24f999
* Content: [_CM_PARTIAL_RESOURCE_LIST (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_partial_resource_list)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_cm_partial_resource_list.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_cm_partial_resource_list.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-03-10:
Thanks for the feedback @tkreuzer, I ran your suggestion past the experts and you are absolutely right :)

Would you want to submit a pull request with your change?  To do that, you can click the **Edit** button on the docs.microsoft.com page, and then click the pencil ("edit this file") icon on GitHub.  We'd of course double check before pushing live so no worries.

If not that's cool too, I'll endeavor to add it soon

tkreuzer, 2020-03-11:
I'm happy to create a PR. WRT the CM_RESOURCE_LIST: Could you provide information regarding the alignment requirement for CM_PARTIAL_RESOURCE_LIST? And is this also a requirement for the size of a variable sized CM_PARTIAL_RESOURCE_DESCRIPTOR or does alignment need to be manually applied to get to the next CM_PARTIAL_RESOURCE_DESCRIPTOR in a CM_RESOURCE_LIST?

tedhudek, 2020-03-12:
There are no special requirements for alignment of the buffer allocation for CM_PARTIAL_RESOURCE_LIST. The only requirement we enforce (already present in the header) is that we use pshpack4.h to force 4-byte alignment of fields in CM_PARTIAL_RESOURCE_DESCRIPTOR, but with any compiler that respects the #pragma it should work without writing extra code.

There is no requirement for the size of the variable, as we already have a requirement that a resource descriptor of type DeviceSpecific must be at the end. 

One caveat (listed in the header) is that there can only be *1* DeviceSpecificData block. It must be located at the end of all resource descriptors in a full descriptor block.  I just pushed a change to add that to the Remarks in https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_partial_resource_descriptor

tedhudek, 2020-03-17:
I added a note to the remarks section of https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ns-wdm-_cm_partial_resource_list with the info above.

closed 2020-03-17 by tedhudek