MicrosoftDocs/windows-driver-docs-ddi #771: IOCTL_DISK_GET_DRIVE_GEOMETRY page looks a bit off

csm10495, 2020-03-13:
I'm 'cheating' a little bit here since I'm not quite sure how to tell you guys this.... The page here: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntdddisk/ni-ntdddisk-ioctl_disk_get_drive_geometry looks fine.

The page here: https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_disk_get_drive_geometry

(which is the same data more/less) doesn't appear to be rendering correctly, like input buffer, output buffer, etc are showing empty on the webpage.

If this isn't the right place, can you redirect this feedback internally to the correct team? ...since a lot of pages are doing this (like for example: https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_disk_get_drive_geometry_ex, https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_disk_is_writable, https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_storage_manage_data_set_attributes, https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_storage_read_capacity, etc...

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 469de7a0-148b-6fe1-0e6c-a5a6fc853a73
* Version Independent ID: 32c1e9eb-5b64-fd66-cfe5-3cc0d3104eb7
* Content: [IOCTL_DISK_GET_DRIVE_GEOMETRY (ntdddisk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntdddisk/ni-ntdddisk-ioctl_disk_get_drive_geometry)
* Content Source: [wdk-ddi-src/content/ntdddisk/ni-ntdddisk-ioctl_disk_get_drive_geometry.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntdddisk/ni-ntdddisk-ioctl_disk_get_drive_geometry.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

hsebs, 2020-03-13:
Win32 API Docs Repository is [here](https://github.com/MicrosoftDocs/sdk-api).

csm10495, 2020-03-13:
Yay. I couldn't find it. Any chance you can move this there? Otherwise I can recreate it.

EliotSeattle, 2020-03-13:
Hi @csm10495 
It's a different repo, but we're one big, happy documentation team here. I've assigned this Issue to the correct owner within the win32 docs.
This looks to be an issue with our build system and these particular files, so while the assignee will escalate to the right people, it's not a simple matter of adding the words. We'll figure it out. It just might take time. (he said, setting expectations)
Thanks for bringing this to our attention.
Eliot

csm10495, 2020-03-14:
Sweet. Let me know when it should be correct (or whenever you have some sort of timeline) and I'll be happy to check it out.

I was getting really confused trying to find these definitions but then realized many seemed weird so wanted to reach out. :)

lorihollasch, 2020-08-07:
@csm10495 , it looks like the Win32 doc team updated the page - I'm not seeing any rendering issues. If you have any further feedback on the Win32 pages, feel free to file issues in their repo (link graciously provided in an above comment by @hsebs ).

closed 2020-08-07 by lorihollasch

csm10495, 2020-08-08:
Hey @lorihollasch , can you ask that team to open up the ability to create Github issues on that repo?

lorihollasch, 2020-08-12:
@csm10495 , just reached out to the Win32 team. They indicated that they will open that repo in the future, but are not able to do so at this point. For now, John Kennedy (the manager) indicated that you should feel free to email him directly at jken@microsoft.com, or you can report issues to Windows Dev Docs (Window Docs) on Twitter.