MicrosoftDocs/windows-driver-docs-ddi #778: Can I get information on keyword conversions?

hsebs, 2020-03-18:
I can't find the information about some keywords such as `-description`, `-parameters`, `-ioctlparameters` and etc. 

Each keyword is displayed differently depending on the situation, but I don't have any information about it.

Can I get information?

EliotSeattle, 2020-03-18:
Hi @hsebs 

Thanks for submitting this Issue. Can you point me to the page (URL) that has the missing descriptions? 

In the future, can you please click the link that says this: 

     You may also leave feedback directly on GitHub .

That link populates a form for the issue with a lot of helpful information for us.
Thanks,
Eliot

hsebs, 2020-03-18:
@EliotSeattle 
I'm sorry to that I wrote this issue here. There was no place to ask about it because [sdk-api](https://github.com/MicrosoftDocs/sdk-api) repository doesn't support issue.

I tried to update [IOCTL_DISK_GET_DRIVE_GEOMETRY control code](https://docs.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_disk_get_drive_geometry) in winioctl.h created from [IOCTL_DISK_GET_DRIVE_GEOMETRY IOCTL](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntdddisk/ni-ntdddisk-ioctl_disk_get_drive_geometry) in ntddk.h, but the `-parameters` section is not visible, and the major code section is still visible.

It seems like `construct-type` problem but I do not know how it is decided.

hsebs, 2020-03-21:
https://github.com/MicrosoftDocs/sdk-api/pull/144#issuecomment-601859994

closed 2020-03-21 by hsebs