MicrosoftDocs/windows-driver-docs-ddi #780: Show me the data!

BrianCatlin, 2020-03-19:
How is the information accessed in the dump file from WinDBG?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1019b134-63f1-320f-bac3-18134891d8bf
* Version Independent ID: 37d47f97-9b24-e600-2c50-da3a6219f927
* Content: [KeAddTriageDumpDataBlock function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-keaddtriagedumpdatablock#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-keaddtriagedumpdatablock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-keaddtriagedumpdatablock.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

grahamm78, 2020-06-19:
Hi Brian,
The information is added as regular VA / memory in the dump, so you use typical commands like dt, dd, etc.

BrianCatlin, 2020-06-19:
Great. WHERE is it located? How is it found?

grahamm78, 2020-06-19:
The crashdmp code / debugger handles all of that for you - in a kernel minidump it's kept in the minidump format with blocks (run .dumpdebug to see the ranges, you will see the ranges you added in the list (perhaps coalesced with adjacent ranges).   In a kernel dump the pages are described by the bitmap dump (the page(s) containing your requested range will be in the dump.  

There is metadata held about the range lists themselves stored as secondary data by the crashdmp driver, so that when the debugger is used to 'carve' a minidump from a kernel/full dump, the set of ranges is read and those ranges are added to the minidump as well.  This can be replicated with ".dump /m" and is also how Windows Error Reporting creates a minidump after a crash where a kernel dump was written.

BrianCatlin, 2020-06-19:
Thanks. That info needs to be in the docs, with a series of examples

grahamm78, 2020-06-19:
Thanks for the feedback, Brian. I'll work with docs team. Which part was
the most helpful? Maybe an example of adding a global, sizeof(global), and
then being clear that that global will be accessible from a minidump?  Or
info on how this helps minidunp carving?  I was also thinking of adding
some best practices to writing bugcheck callbacks, like using
MmIsValidAddress and being careful to avoid infinite loops.  (We may also
need to update docs on MmIsAddressValid based on IRQL requirements)

Cheers,
Graham

BrianCatlin, 2020-06-19:
A step-by-step of how to add information from a driver to the dump, and then accessing it in WinDBG with NATVIS is the most needed

DOMARS, 2020-07-22:
One thing to consider is either re-using the [Driver Module Framework](https://github.com/microsoft/DMF) or extending it to demonstrate the addition of data to the mini dump. There is [Dmf_CrashDump](https://github.com/Microsoft/DMF/blob/master/Dmf/Modules.Library/Dmf_CrashDump.md) that may offer some of this functionality. You can see the code that they implemented for the library in [Dmf_Crashdump.c](https://github.com/microsoft/DMF/blob/master/Dmf/Modules.Library/Dmf_CrashDump.c). If we find specific sample code we can use, I can reference that specific code in a debugger command sample that displays that data, to knit everything together.

tedhudek, 2020-07-22:
I added a link to Dmf_CrashDump in the doc page for KeAddTriageDumpDataBlock.  Don, feel free to add a link there to any new debugger doc you add that covers this more in detail.

closed 2020-07-22 by tedhudek