MicrosoftDocs/windows-driver-docs-ddi #789: No info about the ChangeNotificationInputFlags

gokultv-private, 2020-03-21:
Can you please update the documentation for the ChangeNotificationInputFlags? Currently is it left empty.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3404c7e1-1b80-7f1e-9758-3181588cc165
* Version Independent ID: 62233c6c-158d-c8ae-3400-121f8a98a975
* Content: [_USB_TRANSPORT_CHARACTERISTICS_CHANGE_REGISTRATION (usbioctl.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ns-usbioctl-_usb_transport_characteristics_change_registration?source=docs#feedback)
* Content Source: [wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_transport_characteristics_change_registration.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ns-usbioctl-_usb_transport_characteristics_change_registration.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

EliotSeattle, 2020-04-21:
Broken links fixed. 
Still needs description for ChangeNotificationInputFlags member.

lorihollasch, 2021-02-25:
@gokultv-private , thanks for reporting this issue. I've added the missing descriptions in 2c51eadc , which will go live soon. Sorry for the slow response time!

closed 2021-02-25 by lorihollasch