MicrosoftDocs/windows-driver-docs-ddi #791: Broken link & reference suggestion

jamesmistry, 2020-03-22:
Link https://docs.microsoft.com/windows/desktop/api/fwpstypes/ns-fwpstypes-fwps_incoming_values0_ is broken (the correct URL appears to not have a trailing underscore).

It would very useful to link to a page describing the available FWPS_INCOMING_VALUES values per filtering layer from this page (I regularly find the information by other, less direct means).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0a478874-0fb5-c185-81ac-826e52749e41
* Version Independent ID: d4dbd7f7-819d-674c-24ca-1b4c02c413eb
* Content: [FWPS_CALLOUT_CLASSIFY_FN0 (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nc-fwpsk-fwps_callout_classify_fn0)
* Content Source: [wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nc-fwpsk-fwps_callout_classify_fn0.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

EliotSeattle, 2020-04-21:
Fixed the links in source. Should be live shortly.
Still need to look at request for more info.

tedhudek, 2021-02-24:
Thanks for the feedback and sorry it took us so long.  I no longer repro the bad link so hopefully that got fixed along the way.  For your second suggestion, if you'd like to open a PR to a page describing the available FWPS_INCOMING_VALUES values, we'd love to take a look.

closed 2021-02-24 by tedhudek