MicrosoftDocs/windows-driver-docs-ddi #808: Flags not mentioned?

cmusta, 2020-04-08:
There is no mention here of the FWPS_INJECTION_TYPE_L2 or FWPS_INJECTION_TYPE_VSWITCH_TRANSPORT flags.  I'm assuming that the MAC Ethernet and Native MAC layers can use the FWPS_INJECTION_TYPE_L2 flag.  I trying to figure out if I hook both the MAC Ethernet and NATIVE MAC layers, can I use the same handle for both or generate 2 separate handles.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d50157dd-80ff-8049-6f5c-80a081f89e5c
* Version Independent ID: be2d2e88-196b-faa0-04b3-2cf33e057560
* Content: [FwpsInjectionHandleCreate0 function (fwpsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fwpsk/nf-fwpsk-fwpsinjectionhandlecreate0#feedback)
* Content Source: [wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectionhandlecreate0.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjectionhandlecreate0.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

OZ1, 2021-01-07:
any progress?

billlattimer, 2021-01-07:
@OZ1 I'll bump the priority and work to close this out ASAP. Apologies for the delay.

tedhudek, 2021-03-09:
Hi @cmusta, @OZ1 thanks for the feedback and apologies for the delay.

For the MAC layers ( *MAC_FRAME_NATIVE, *MAC_FRAME_ETHERNET, *VSWITCH_ETHERNET), you can use the same injection handle acquired with the FWPS_INJECTION_TYPE_L2 flag.  The Injection type is tied closer to the injection functions than the layers.

See https://github.com/microsoft/Windows-driver-samples/blob/master/network/trans/WFPSampler/syslib/HelperFunctions_InjectionData.cpp for an example.

closed 2021-03-09 by tedhudek