MicrosoftDocs/windows-driver-docs-ddi #815: WskReceiveEvent called in parallel?

DrChat, 2020-04-16:
It'd be nice to have a little blurb in the remarks about whether this function will be called _as soon as_ data is received, even if another instance of the callback is currently running, or if Winsock Kernel will call this function sequentially. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f53191fb-0657-43f7-3501-8e2eefce82d3
* Version Independent ID: 3bddeb26-35a3-d043-a1bb-ee600fac14f7
* Content: [PFN_WSK_RECEIVE_EVENT (wsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wsk/nc-wsk-pfn_wsk_receive_event#feedback)
* Content Source: [wdk-ddi-src/content/wsk/nc-wsk-pfn_wsk_receive_event.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wsk/nc-wsk-pfn_wsk_receive_event.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-03-10:
Thanks for the feedback @DrChat and apologies for the delay.

Winsock Kernel (WSK) calls this callback serially, so it is not always invoked as soon as data is received.

I've clarified this on the doc page - that update should be live within the hour.

closed 2021-03-10 by tedhudek

DrChat, 2021-03-11:
Appreciate it! 👍