MicrosoftDocs/windows-driver-docs-ddi #820: Requesting clarification

pankajbharti, 2020-04-23:
"If a driver that does not own the IRP calls IoCancelIrp, the results are unpredictable. The IRP might be completed with a successful status even though its cancel bit was set."

There is scope for correction here. 

The owner of the irp is the driver which is handling the irp. That driver never calls IoCancelIrp. The sender of the IRP is the one who calls IoCancelIrp. The sender of the Irp is not the owner of Irp. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 00a3f826-e7b3-3063-fc4a-48143bf6ebaf
* Version Independent ID: fb3ab962-9b52-216d-690f-4190ddc2e75c
* Content: [IoCancelIrp function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iocancelirp#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iocancelirp.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iocancelirp.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-04-23:
Thanks @pankajbharti for the feedback. I think I understand what you are saying. Do you have a suggestion for rewording the passage?

tedhudek, 2020-07-22:
Closing out due to inactivity

closed 2020-07-22 by tedhudek