MicrosoftDocs/windows-driver-docs-ddi #821: STATUS_INVALID_PARAMETER returned on cross-volume symlinks on Windows 10

alfaunits, 2020-04-27:
This is contrary to previous OSes where the symlink would be resolved even across volumes.
We have been able to reproduce this only on Windows 10/Server 2016+:
- Map a network drive to Z:\
- Run "mklink /d C:\NetDrive Z:\" from Admin Command Prompt
- Try to create a new file.
A minifilter attached to the C drive would get STATUS_INVALID_PARAMETER for the FGFNI call when normalized path is requested.

This does not happen on previous version, the API works.

Note that this could be a bug in ZwQueryInformationFile or Lanman redirector. Because it is not possible to get the normalized file name via ZwQueryInformationFile in above scenario.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3791685f-ae6e-06d3-d42b-07f498a5e36a
* Version Independent ID: 254f85ed-231d-c3e4-db43-fd13d8150db1
* Content: [FltGetFileNameInformation function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltgetfilenameinformation#feedback)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-07-28:
@alfaunits , I'm guessing that this is an issue with querying normalized file names on a remote volume versus being symlink-related? I did add a NOTE that starting with Windows 10, SMB doesn't support querying normalized file names. Thanks for flagging this.

closed 2020-07-28 by lorihollasch

alfaunits, 2020-08-01:
Hi, Lori. I'm not sure..
Querying normalized file names still works on remote volumes. It also works if the symlink is opened, its target queried, and then a normalized file name queried.
Note that I have been told the query (over symlink) works again in Windows 10 build 2004, but did not confirm this myself.

I think the issue is with SMB not handling FILE_NORMALIZED_NAME_INFORMATION properly when a remote file is opened via a symlink from a local volume, as demonstrated in my example.
I noticed there is added text to the article:
"Starting in Windows 10, you cannot query normalized file names on a remote volume because the Server Message Block (SMB) Protocol does not support queries for normalized paths."
This is not correct (generally, the query still works correctly, just not in all cases). Rather, STATUS_INVALID_PARAMETER should be documented to be a possible error code when resolving remote file names, if a symlink from another volume points to them on Windows 10/Server 2016.