MicrosoftDocs/windows-driver-docs-ddi #826: Is the buffer writeable

rodwiddowson, 2020-05-05:
I'll do the test myself, but the whole construction is meaningless to a Filter if it cannot write the buffer (to reflect the work it does in the background).

I'm assuming it is, but it might be good to have it recorded.

Also there is no *available since* field.

As an aside it isn't obvious to me why this function is needed.  Just make the ECP scrutable (which it is) and the structure unchanging (which it should be).  But this is not my product so all I can do is draw conclusions, not make recommendations

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c49e9cf3-1951-0a50-8a29-422846166e64
* Version Independent ID: 69cc5ab8-8674-23ed-de8e-b5b02a57291e
* Content: [FltRetrieveFileInfoOnCreateCompletionEx function - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltretrievefileinfooncreatecompletionex)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltretrievefileinfooncreatecompletionex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltretrievefileinfooncreatecompletionex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2020-05-07:
Empirically the value is indeed the ECP buffer plus the appropriate (per info type, per OS version) offset.

lorihollasch, 2021-03-12:
@rodwiddowson , I was finally able to get this page updated. Added details about the buffer being writable, the version info, why the function is useful (ECP is "opaque" here), along with some other technical goodies gleaned from Christian. I also added pages for the missing QoC structures. Thanks for reporting this. Apologies for the slow turn around!

closed 2021-03-12 by lorihollasch