MicrosoftDocs/windows-driver-docs-ddi #832: API DeviceIoControl with this control code failed

Ouqingjun, 2020-05-16:
for some os win10, API DeviceIoControl with this control code failed, when i try to get the serialNum string for a usb device, and i get a error code 0 by GetLastError(), if i need do some special configuration ? thanks

But the same code works on win7.

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: 5dc6187b-b3fb-f036-1733-4bc9443d4fe1
* Version Independent ID: 7178a780-b2d7-30f2-b510-11d064e26511
* Content: [IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION (usbioctl.h) - Windows drivers](https://docs.microsoft.com/zh-cn/windows-hardware/drivers/ddi/usbioctl/ni-usbioctl-ioctl_usb_get_descriptor_from_node_connection#feedback)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_descriptor_from_node_connection.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_descriptor_from_node_connection.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-03-11:
Hi @Ouqingjun thanks for the feedback and apologies it's taken us so long.  Your question is now tracked in a Microsoft-internal bug.  I do not have an ETA for when it will be resolved.  Because we have the bug tracking it now, I am going to close out this issue.

If you would like me to notify you if and when I get an answer, please just reply to me here and I will see it and make a note.

Thanks for caring about the docs!

closed 2021-03-11 by tedhudek

Ouqingjun, 2021-03-25:
dear, it is so suprised for me to get a feedback, please notify me when this issue be resolved.

thanks and best wishes for you!