MicrosoftDocs/windows-driver-docs-ddi #838: RtlEqualMemory *does* return a value

rodwiddowson, 2020-05-25:
Unfortunately this is not fixable in the markup (I looked and the ReturnValue paragraph appears to be automatically generated).

The "remarks" section contains the truth but I had a hiccough when the [section above](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtlequalmemory?redirectedfrom=MSDN#return-value) says "No value".

Its probably because this is in fact a (reasonably weird) macro and is unfixable, but if it is fixable it would be good.

Be safe

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d66bcef4-c694-d12e-b535-90f2aec47748
* Version Independent ID: 3e6a4e85-ee54-baa4-fd08-7b04fe0bb6e2
* Content: [RtlEqualMemory macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtlequalmemory?redirectedfrom=MSDN)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlequalmemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlequalmemory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-05-28:
@aviviano: This needs a ## -returns section.  We should also move the type info out of the parameter descriptions up into the (already manually overridden) syntax block, to match what's in the header.

rodwiddowson, 2020-05-28:
@tedhudek thats cool. I guess I missed the language capabilities of this markup.  It there a ‘Fine Manual’ I can skim for my future reference?

Thanks

tedhudek, 2020-05-28:
Appreciate the interest, Rod; there isn't really anything I can point you to. This repo gets integrated with the auto-parsed headers and the merged result is what gets published. Hence the disconnect sometimes between the page you see on the docs site and what's in this repo.

We can override the auto-generated syntax block if it gets it wrong, as we did for the page you found here. Otherwise `## -syntax` doesn't need to be in there; that block is auto-generated and then parm/member names get added next to the matched up description name. 

The upside is that it enforces fidelity with the headers, except when it messes up :)

aviviano, 2020-06-08:
Thanks @rodwiddowson. The fix is up now.

closed 2020-06-08 by aviviano