MicrosoftDocs/windows-driver-docs-ddi #847: Duplicate entries in Help Viewer

PisarevSergey, 2020-06-07:
Duplicate entries for many functions in local Help viewer

[image]

#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f2c03664-482d-34f8-ac15-e145ca2987a6
* Version Independent ID: e6bb02c6-7e6d-d24d-386c-94959d652c22
* Content: [FltGetVolumeProperties function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/nf-fltkernel-fltgetvolumeproperties)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumeproperties.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumeproperties.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

tedhudek, 2020-06-09:
Thanks @PisarevSergey for bringing this to our attention. Is it possible that you have both the Windows Driver Development book *and* the WDK book installed in your offline viewer?  If so, you should remove the Windows Driver Development one.  That one is out-of-date and has been superseded by the WDK one.

PisarevSergey, 2020-06-09:
Thank you!
Maybe it’s better to remove old book from help viewer completely? So that it didn’t appear in the list of available books. Or at least mark it as deprecated ?

tedhudek, 2020-06-09:
I didn't want to comment because I hadn't confirmed it, but I think it got removed this morning, thanks to your nudge!

closed 2020-06-09 by tedhudek