MicrosoftDocs/windows-driver-docs-ddi #855: not searchable in Help Viewer by index

PisarevSergey, 2020-06-12:
Present in offline help (Help Viewer 2.3 for Visual Studio 2019) but not searchable in index. Can be found through search tab.

[image]

[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a39b4d45-43aa-5a0c-9a37-fb0e32beb698
* Version Independent ID: 6d767c49-f9e4-2fe1-68b6-c3d24fe61ee9
* Content: [_FLT_CALLBACK_DATA (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/fltkernel/ns-fltkernel-_flt_callback_data)
* Content Source: [wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_callback_data.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_callback_data.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-07-29:
@PisarevSergey , the offline books were updated in June. Are you still seeing this issue?

lorihollasch, 2020-07-31:
Closing this as I believe this issue was fixed. @PisarevSergey , let me know if this is still an issue for you, thanks.

closed 2020-07-31 by lorihollasch

PisarevSergey, 2020-07-31:
Unfortunately the problem still persists. I have freshly installed windows, so everything is definitely up to date

tedhudek, 2020-08-04:
It is actually in the index, but it has an underscore prepended. So the index entry is `_FLT_CALLBACK_DATA`.  The front of the index has all of the entries starting with underscore.

lorihollasch, 2020-08-04:
I just removed and then re-added the WDK books under "Manage Content", and got notification of several updated during the Add. I can now discover _FLT_CALLBACK_DATA in an Index search as Ted indicates.

PisarevSergey, 2020-08-05:
I understand that documentation automatically generated from include file(correct me if I’m wrong).
So I can see why _FLT_CALLBACK_DATA considered a correct spelling since this is “true” name of the type.
However non-underscore version actually used. Always. 
Such typedef is a classic C trick so that you don’t need to prepend type name with struct and just use name itself.

Maybe index generating algorithm can be changed ?

tedhudek, 2020-08-06:
Working on it :)

tedhudek, 2020-08-06:
Ok Sergey, I just updated the WDK ref book.  I tested it and `FLT_CALLBACK_DATA` now appears under F in the index.

Please remove it from your Help Viewer and then add it again to download the most recent.

PisarevSergey, 2020-08-07:
Thank you !
Works perfectly after update.
Didn’t even need to remove book explicitly