MicrosoftDocs/windows-driver-docs-ddi #859: KeExpandKernelStackAndCalloutEx missing

rodwiddowson, 2020-06-12:
I just realized also 

- KeExpandKernelStackAndCalloutEx is missing from the docset (and not at the obvious place)
- all the links in this page to KeExpandKernelStackAndCalloutEx are stale

(plus of course the xommenrs I made in #858 are also relevant.

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9ad3b97d-fac8-9f5a-9f6d-a49818f2944a
* Version Independent ID: 796528a1-d67e-449d-9077-35cf6b2de0c9
* Content: [KeExpandKernelStackAndCallout function (ntddk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/nf-ntddk-keexpandkernelstackandcallout)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-keexpandkernelstackandcallout.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-keexpandkernelstackandcallout.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-12:
Thanks that's a good find. It looks like the text of the page is still good on previous-versions, if you need to refer to it.  Meanwhile I'll work on restoring it to the regular site.

tedhudek, 2020-06-18:
I just pushed KeExpandKernelStackAndCalloutEx.  Should be live soon.

KalleOlaviNiemitalo, 2020-07-15:
The KeExpandKernelStackAndCalloutEx link on the KeExpandKernelStackAndCallout page seems to be working now.

However, the "See also" section of the KeExpandKernelStackAndCallout page contains a link to KeExpandKernelStackAndCallout itself, not to KeExpandKernelStackAndCalloutEx. That doesn't seem right.

tedhudek, 2020-07-15:
Thanks @KalleOlaviNiemitalo 

Just pushed this fix.  Should be live soon!

closed 2020-07-15 by tedhudek