MicrosoftDocs/windows-driver-docs-ddi #860: link in Help Viewer 2.3 leads to online library and not local

PisarevSergey, 2020-06-12:
[image]

should lead to

[image]

but it leads to online MSDN library

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9654d4be-9f3d-444a-97df-0c2563c4adc7
* Version Independent ID: 98b7ca6b-0067-82be-98f0-c8d96148e6fd
* Content: [_POOL_TYPE (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_pool_type)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_pool_type.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_pool_type.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-12:
In the source, that link is:

`<a href="https://docs.microsoft.com/windows-hardware/drivers/kernel/no-execute-nonpaged-pool">No-Execute (NX) Nonpaged Pool</a>`

Maybe a full URL like that always opens a browser page?  Not sure there's anything we can do about that.

PisarevSergey, 2020-06-13:
In general links in Help Viewer lead to another pages in offline help. So it is definitely possible( and I  would think even expected) for offline help to be self-contained. Maybe this link should be changed to point to offline content ?

tedhudek, 2020-06-16:
This affects links that go from the "conceptual" (non-API) docs to the API docs, or from API docs to conceptual docs.  I'm investigating what we can do about it.

tedhudek, 2020-06-17:
After investigating, I am pretty sure that it is not currently possible to link from a page in one offline book to a page in a different offline book. I looked at several different link formats. Nor is it currently possible to build a single offline book that has these two (conceptual and reference) TOC trees.

Do you have any examples of cross-book links that work?

PisarevSergey, 2020-06-18:
I don’t have examples of cross-book references. It is understandable that one book can’t link another.
Maybe in the future it will be possible to merge kernel documentation in single book ?

tedhudek, 2020-06-18:
I will keep that in mind.

Keeping this issue open because I am attempting to publish the Win10 v2004 updated docs to the offline viewer, but we've hit a snag. I'll update here when those are available.

tedhudek, 2020-07-02:
Thanks again for the help and testing on this @PisarevSergey

closed 2020-07-02 by tedhudek