MicrosoftDocs/windows-driver-docs-ddi #861: Some of the formatting is messed up.  Objects and Resources table, and the one below it. These do not appear as a table but as markup

NathanS-Microsoft, 2020-06-12:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f0a69c83-8e5a-df94-2fb6-c2e19bc2b60a
* Version Independent ID: 6de870f8-62d4-c6a8-1b70-69b5276437e0
* Content: [Windows kernel - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/_kernel/)
* Content Source: [wdk-ddi-src/content/_kernel/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_kernel/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2020-06-12:
Thanks for spotting this and submitting feedback, @NathanS-Microsoft 
I can own this cleanup on this page moving forward. @tedhudek, feel free to help, too. :-)
Eliot

EliotSeattle, 2020-06-16:
Thanks, again, @NathanS-Microsoft 

I believe that I have cleaned up all the malformed content. Please feel free to let me know if you spot any other errors.

I noticed missing links ("For more info, see..."). I'll take care of them.

More important to me is the question of whether a page of this enormity is useful, even with the quick links on the right nav. 

I'm going to close this particular Issue, as the request is fulfilled, but if you have any thoughts about the page, please reach out to me through GitHub or in email.

Appreciate your help.

Eliot

closed 2020-06-16 by EliotSeattle

NathanS-Microsoft, 2020-06-16:
It was useful to me, sort of a one-stop shopping of the driver interface.  But yeah, it’s a long page, well-suited to break it up.