MicrosoftDocs/windows-driver-docs-ddi #867: Visual Studio F1 should not point to previous version docs

NextTurn, 2020-06-15:
Sorry if wrong place. In Visual Studio if you click on a function which is supported for both NDIS 6.0 and NDIS 5.1, the NDIS 5.1 page will be opened.

Take `NdisAllocateSpinLock` as an example:

Query: https://msdn.microsoft.com/query/dev16.query?appId=Dev16IDEF1&l=EN-US&k=k(NDIS%2FNdisAllocateSpinLock);k(NdisAllocateSpinLock);k(DevLang-C%2B%2B);k(TargetOS-Windows)&rd=true

Actual: https://docs.microsoft.com/previous-versions/windows/hardware/network/ff550793(v=vs.85)?redirectedfrom=MSDN

Expected: https://docs.microsoft.com/windows-hardware/drivers/ddi/ndis/nf-ndis-ndisallocatespinlock

aviviano, 2020-07-09:
Thanks for letting us know about this issue @NextTurn. We'll have a look and get back to you.

tedhudek, 2020-07-13:
I just pushed a change that I think will fix this problem.  The fix should appear on the live site within an hour.

tedhudek, 2020-07-13:
@NextTurn can you please try again?

NextTurn, 2020-07-14:
@tedhudek I found that the [`NdisInterlockedInsertTailList`](https://msdn.microsoft.com/query/dev16.query?appId=Dev16IDEF1&l=EN-US&k=k(TAP%2FTAP_ADAPTER%3A%3ANdisInterlockedInsertTailList);k(TAP_ADAPTER%3A%3ANdisInterlockedInsertTailList);k(NdisInterlockedInsertTailList);k(DevLang-C%2B%2B);k(TargetOS-Windows)&rd=true) macro function still points to the old page.

tedhudek, 2020-07-15:
Hi @NextTurn,

I just pushed a test fix for NdisAllocateSpinLock and NdisInterlockedInsertTailList.  Please wait a couple hours and try again with those two functions.  Apologies for the inconvenience, and thank you for helping us test the fix.  If it works, I'll make the same metadata change across other pages as well.

NextTurn, 2020-07-16:
The links are working as expected, thanks!

tedhudek, 2020-07-17:
Thanks.  I've got a fix ready that updates all 22k API pages in this repo.  It's testing now, hoping to push it live tomorrow.

tedhudek, 2020-07-17:
I just pushed the change.  The fix should be live within the hour.  Thanks again for reporting this problem!

closed 2020-07-17 by tedhudek