MicrosoftDocs/windows-driver-docs-ddi #868: Duplicate entries in navigation panel for "Engextcpp.h"

matwilli, 2020-06-18:
Navigation panel reads "Debugger", "Dbgeng.h", "Dbgmodel.h", "Engextcpp.h", **"Engextcpp.h"**, "Extsfns.h", "Wdbgexts.h"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 24f0d91d-43e6-b2e2-aa4c-d56cbe5c204a
* Version Independent ID: 43e551b6-3ccd-7c7e-b33a-08f36cc2a0e7
* Content: [Debugger - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/_debugger/)
* Content Source: [wdk-ddi-src/content/_debugger/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/_debugger/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2020-06-22:
@matwilli - Investigating - Looks like our publishing system may be tripping over the .hpp file extension.

DOMARS, 2020-06-23:
@matwilli - The second, duplicate entry for engextcpp has been removed from the navigation panel. Thanks for seeing this error and for filing this issue.

closed 2020-06-23 by DOMARS