MicrosoftDocs/windows-driver-docs-ddi #870: NTSTATUS codes are documented in Markdown but not shown at docs.microsoft.com

KalleOlaviNiemitalo, 2020-06-19:
The Markdown source code of the IOCTL_REDIR_QUERY_PATH documentation contains a list of recommended NTSTATUS codes such as STATUS_BAD_NETWORK_NAME, but the resulting HTML at docs.microsoft.com omits this list entirely.

The documentation of [IOCTL_REDIR_QUERY_PATH_EX](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ni-ntifs-ioctl_redir_query_path_ex) and [IOCTL_VOLSNAP_FLUSH_AND_HOLD_WRITES](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ni-ntifs-ioctl_volsnap_flush_and_hold_writes) has the same problem.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: baa1ac47-0260-065f-5406-bdf4db440e4d
* Version Independent ID: c23b4b32-0c7e-cad2-44b2-671dc536d196
* Content: [IOCTL_REDIR_QUERY_PATH (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ni-ntifs-ioctl_redir_query_path)
* Content Source: [wdk-ddi-src/content/ntifs/ni-ntifs-ioctl_redir_query_path.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ni-ntifs-ioctl_redir_query_path.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2020-06-19:
Thanks, @KalleOlaviNiemitalo 
Working on it.

tedhudek, 2020-06-19:
This is a great find.  I just pushed a manual fix for those 3 files and am working on an automated one for 44 additional affected files

tedhudek, 2020-06-19:
Just pushed the change for the rest of the occurrences.  Thanks again @KalleOlaviNiemitalo!

closed 2020-06-19 by tedhudek

KalleOlaviNiemitalo, 2020-06-19:
"the above list" seems wrong as the list is actually below. Likewise "such as one of the following:" when the list does not follow immediately.