MicrosoftDocs/windows-driver-docs-ddi #882: Vanished from Help Viewer 2.3

PisarevSergey, 2020-06-23:
After recent updates vanished from offline help
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0f30189f-18df-1474-4bd0-84dfa6f73e22
* Version Independent ID: beb5b914-9f72-e25b-7c3f-3b47bb5973eb
* Content: [ZwCreateFile function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwcreatefile)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwcreatefile.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwcreatefile.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-23:
AFAIK we haven't republished the offline books yet.  Can you provide detail on how you updated your offline books and what you're seeing? Sorry this hasn't been the smoothest path

PisarevSergey, 2020-06-23:
This is my installed books:

[image]

ZwCreateFile topic cannot be found through both Index and Search tabs

tedhudek, 2020-06-26:
I'm tracking this issue separately with the team that manages our offline books, so closing this out.  Appreciate the report

closed 2020-06-26 by tedhudek

tedhudek, 2020-06-29:
Just pushed a fix for this btw Sergey - will need to rebuild offline books for you to see the fix

tedhudek, 2020-06-30:
Update - I just updated the "Windows Driver Kit(WDK)" offline book.  Note- I have not yet updated the WDK ref book.  Let me know if it looks as expected for you

PisarevSergey, 2020-06-30:
Have installed documentation update. Still no zwcreatefile topic in help viewer - this is expected since ref book not updated, right ?
Otherwise everything looks the same

tedhudek, 2020-06-30:
Right.  The top node should have what's new for Win10 v 2004

tedhudek, 2020-06-30:
I just pushed the new ref book. Try removing the one you have, re-adding from online, and then search for ZwCreateKey.  It's back!

It's a known issue that there are duplicate index entries for many pages.  This is because of the TOC behavior described at https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/

PisarevSergey, 2020-06-30:
Didn't remove the old book, just installed the update. 
ZwCreateFile now searchable through Search and Index tabs.

Is it actually required/ recommended to remove old book before update ?

tedhudek, 2020-06-30:
I think you might get two copies of each page if you have two versions of the same book installed.  But if it works for you, then it's good :)

PisarevSergey, 2020-07-01:
I removed and reinstalled all documentation. 
Looks the same so far. 
I'm getting one result for ZwCreateFile in Index, but 2 results in Search - both point to the same page.
But I guess this is not a big problem, I almost exclusively use Index and very rarely Search.