MicrosoftDocs/windows-driver-docs-ddi #884: Unhelpful link "WSK Address Families"

NextTurn, 2020-06-24:
The three "WSK Address Families" links point to [the previous version of Ws2def.h](https://docs.microsoft.com/previous-versions/windows/hardware/drivers/mt808757(v=vs.85)) which doesn't tell me anything about address families.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9e1789e8-4d32-116e-6c57-ff801546989e
* Version Independent ID: e02f1cd1-cbef-6960-bc8a-017cca50bccf
* Content: [PFN_WSK_SOCKET (wsk.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wsk/nc-wsk-pfn_wsk_socket#feedback)
* Content Source: [wdk-ddi-src/content/wsk/nc-wsk-pfn_wsk_socket.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wsk/nc-wsk-pfn_wsk_socket.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2020-06-24:
Thanks @NextTurn for bringing this to our attention. We'll look into this and get back to you.

aviviano, 2020-11-03:
Thank you @NextTurn for your excellent feedback. We've updated this reference to WSK Address Families to link to https://docs.microsoft.com/en-us/windows-hardware/drivers/network/ws2def-h.

closed 2020-11-03 by aviviano