MicrosoftDocs/windows-driver-docs-ddi #904: Incorrect return type

AzAgarampur, 2020-07-12:
According to the header file `wdm.h` using the latest WDK and Windows 10 SDK, this function returns type `int` instead of `NTSTATUS`. However, I assume the `int` is to be treated the same as an `NTSTATUS` value.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a4f71c4-5e16-4662-4e62-d6a19837ffc9
* Version Independent ID: 91fbf4d7-4008-6435-37cf-8b5376eacdbb
* Content: [RtlIntegerToUnicodeString function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-rtlintegertounicodestring)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlintegertounicodestring.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlintegertounicodestring.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-07-14:
Hi @AzAgarampur, thanks for the report.  On line 12013 of wdm.h (Vb WDK), I see:

```cpp
#if (NTDDI_VERSION >= NTDDI_WIN2K)
_IRQL_requires_max_(PASSIVE_LEVEL)
_At_(String->MaximumLength, _Const_)
NTSYSAPI
NTSTATUS
NTAPI
RtlIntegerToUnicodeString (
    _In_ ULONG Value,
    _In_opt_ ULONG Base,
    _Inout_ PUNICODE_STRING String
    );
#endif
```

What build is your kit from?

AzAgarampur, 2020-07-17:
Hey @tedhudek,

Sorry for the confusion. I found out that the installer was terminated during an install, leaving partial header files. VS's intellisense was showing the return type as a guessed `int` type.

So there's no problem anymore, it was on my end. Sorry for the confusion.

tedhudek, 2020-07-17:
Got it!  Thanks for the update, and let us know if we can help further :)

closed 2020-07-17 by tedhudek