MicrosoftDocs/windows-driver-docs-ddi #913: How to check which object was signaled, when calling KeWaitForMultipleObjects with WaitAny

KalleOlaviNiemitalo, 2020-07-15:
If a driver calls **KeWaitForMultipleObjects** with *Count* > 1 and *WaitType* = **WaitAny**, then the call will return when any of the objects is signaled (or the wait times out etc.). How can the driver know which object was signaled?

If I remember correctly, **KeWaitForMultipleObjects** indicates that by returning **STATUS_WAIT_0** + *index* or **STATUS_ABANDONED_WAIT_0** + *index*, where 0 ≤ *index* < *Count*. However, the current documentation does not mention that.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bfc572c9-2cbb-0822-ba10-c29975752306
* Version Independent ID: 117995e4-b3cd-b56c-dbda-8a2acd4e2b46
* Content: [KeWaitForMultipleObjects function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-kewaitformultipleobjects)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-kewaitformultipleobjects.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-kewaitformultipleobjects.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

KalleOlaviNiemitalo, 2020-07-15:
Microsoft Windows Driver Kit Documentation 6001.071220 documents those return values, but I don't think the license terms allow me to copy the text to this comment.

tedhudek, 2020-07-15:
Hi @KalleOlaviNiemitalo 

Thanks for the question.  You are exactly right, it works just like it did ~10 years ago. Thanks for the tip on the old docs, I was able to retrieve the old text from there.  I just pushed the fix - should be live within an hour.

closed 2020-07-15 by tedhudek