MicrosoftDocs/windows-driver-docs-ddi #914: KEY_WRITE constant in DesiredAccess table has incorrect spelling of flag KEY_CREATE_SUB_KEY

LCBradley3k, 2020-07-21:
Notice how under "DesiredAccess flag" table we see the flag "KEY_CREATE_SUB_KEY".

Now if you look at the list of flags for constant "KEY_WRITE", you'll notice it contains "KEY_CREATE_SUBKEY". This should be spelt "KEY_CREATE_SUB_KEY". It is missing an underscore between SUB and KEY.

Looking at the source code I have confirmed it is "KEY_CREATE_SUB_KEY" and not "KEY_CREATE_SUBKEY".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a13506d3-e14e-0633-1703-4899c44e8b6f
* Version Independent ID: 5aa9e3f6-5d20-a000-26da-99c65f2601c3
* Content: [ZwCreateKey function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-zwcreatekey)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-zwcreatekey.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-zwcreatekey.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

LCBradley3k, 2020-07-21:
Adding the pull request I made: #915

tedhudek, 2020-07-21:
Thanks again!

closed 2020-07-21 by tedhudek