MicrosoftDocs/windows-driver-docs-ddi #917: NtQueryInformationFile: HTML is mangled

maharmstone, 2020-07-25:
The HTML is mangled in the middle of this page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 65a6b1c6-aeaf-e922-b8ea-47d7469e96a4
* Version Independent ID: 0cbfe627-8995-97eb-185c-06e3fa202bbe
* Content: [NtQueryInformationFile function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntqueryinformationfile)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationfile.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntqueryinformationfile.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2020-07-27:
Thanks! Looking into this.

EliotSeattle, 2020-07-27:
Fixed in the source. Should go live later today.

Thanks for the heads up, @maharmstone

closed 2020-07-27 by EliotSeattle

KalleOlaviNiemitalo, 2020-08-04:
That's much better! The description of FileAttributeTagInformation is still in the wrong column, though.

EliotSeattle, 2020-08-04:
D'OH!

Thanks for reviewing, @KalleOlaviNiemitalo. I'm embarrassed that I didn't proofread after the changes. I fixed the extra pipe symbol--and verified the fix in preview. Checked into the main branch. Will be live shortly.

Cheers,
Eliot