MicrosoftDocs/windows-driver-docs-ddi #921: _FILE_STAT_INFORMATION structure refers to nonexistent Remarks section

KalleOlaviNiemitalo, 2020-08-04:
The _FILE_STAT_INFORMATION article says:

> `ReparseTag`
>
> Reparse point tag. Must be a Microsoft reparse point tag. (See the following **Remarks** section.)

However, the article does not have a Remarks section.

A Remarks section could describe what the value of the ReparseTag field will be if a driver calls [NtQueryInformationByName](https://docs.microsoft.com/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntqueryinformationbyname) with FileStatInformation and the file has a non-Microsoft reparse tag. (The [NtQueryInformationFile](https://docs.microsoft.com/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntqueryinformationfile) and [NtSetInformationFile ](https://docs.microsoft.com/windows-hardware/drivers/ddi/ntifs/nf-ntifs-ntsetinformationfile) articles do not list FileStatInformation as a valid information class.)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7ab6def9-2aa0-ed56-61c3-97670d2de50f
* Version Independent ID: 0f19f3dd-62de-caca-e94b-ae868d31ea60
* Content: [_FILE_STAT_INFORMATION (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntifs/ns-ntifs-_file_stat_information)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_file_stat_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_file_stat_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

hsebs, 2020-10-24:
FILE_STAT_LX_INFORMATION page has same problem.

lorihollasch, 2020-11-25:
@KalleOlaviNiemitalo , thanks for this PR and apologies for my slow response. Both this page and the LX page (thanks @hsebs ) were indeed wrong about reparse point tags - they don't have to be Microsoft tags. I've updated both pages. I've also updated NtQueryInformationFile and NtQueryinformationByName to have up-to-date FILE_INFORMATION_CLASS lists.

closed 2020-11-25 by lorihollasch