MicrosoftDocs/windows-driver-docs-ddi #925: Members of IDebugEventContextCallbacks are missing

NextTurn, 2020-08-09:
Especially the usage of the additional `Context` and `ContextSize` arguments should be explained.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 61114cba-7b34-bb75-c63b-ffaf9d180856
* Version Independent ID: 3b75adc7-728f-8e35-2ea8-b67fd4c8e53a
* Content: [IDebugEventContextCallbacks (dbgeng.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/dbgeng/nn-dbgeng-idebugeventcontextcallbacks)
* Content Source: [wdk-ddi-src/content/dbgeng/nn-dbgeng-idebugeventcontextcallbacks.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgeng/nn-dbgeng-idebugeventcontextcallbacks.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-01-25:
@nxtn - Investigating this git docs issue - I see what you are saying.

DOMARS, 2021-02-18:
@nxtn - I worked with the dev team to document the primary differences with  IDebugEventContextCallbacks in the dbgeng header and add 16 new topics. IDebugEventContextCallbacks  is similar to to the IDebugEventCallbacks interface, but this interface includes a [DEBUG_EVENT_CONTEXT structure](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/dbgeng/ns-dbgeng-_debug_event_context) as the “context” parameter of each event callback. So in some cases getting that additional context back may be handy, depending on what the goals of your code are and which callback you are working with. Thanks for your interest in Windows debugging code development, and thanks for filing this issue.

closed 2021-02-18 by DOMARS